Lora-net/lora_gateway #155: Debugging lgw_send (not sending)

wavesoft, 2019-12-30:
Hi All!

I have ported `libgw` and fragments of `lora_pkt_fwd` in order to be able to control a RAK2247 concentrator from an MCU. I can say that I have read through about 80% of the library by now and my only unknown parts are the internals of SX1301.

So far I had great success with the API abstractions and I am successfully receiving uplink packets and forwarding them to my server. However, I have some issues with my downlink that I cannot explain.

In detail:
* I receive an uplink packet with `"tmst":666233252` from the concentrator
* I forward it to the server and I receive a timelink packet with `"tmst":671233252`
* I call `lgw_send` passing down the parameters received from the server and I notice a successful scheduling of the transmission:
```
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
Info: packet will be sent without CRC
INFO: Enabling TX notch filter
lgw_send done: count_us=671233252
```
* Checking the device, I *never* see an incoming downstream packet

Moreover, in future transmissions (a few minutes later) I see the following warning, that obviously indicates that SX1301 is somehow still in "Transmission" mode
```
a downlink was already scheduled, overwritting it...
```

## Question

Why isn't the transmissions happening? How can I debug _why_ it's not leaving the concentrator?

## Debug Info

* I have configured the `libloragw` with the values from `global_conf.json.PCB_E286.EU868.basic`.
* The contents of status register before the **first** transmission is `LGW_TX_STATUS=80`
* The contents of the status register before the **second** transmission is `LGW_TX_STATUS=91 `
* I noticed that the `LGW_TX_STATUS` resets to `80` after ~70 minutes which is suspiciously close to the time the 32-bit microsecond-timer overflows.
* The actual payload that goes to `LGW_TX_DATA_BUF_DATA` is:
```
0x3ffee2c0 59 06 66 08 b9 d7 ec 0a 00 17 21 10 00 08 00 00 |Y.f.......!.....|
0x3ffee2d0 20 53 b2 ab 8e 7f 2f d2 20 63 50 a7 8e aa e1 82 | S..../. cP.....|
0x3ffee2e0 4e 55 65 38 f6 70 2f e8 bc 65 14 3e 60 66 2c eb |NUe8.p/..e.>`f,.|
0x3ffee2f0 91 |.|
```

wavesoft, 2019-12-31:
To my utter surprise, it ends up to be an SPI issue!!!

closed 2019-12-31 by wavesoft

lichuanhe, 2020-06-02:
hello，the same problem comes  2me ,when the node send package with ack. sometimes the device cannot receie ack,then ,the the gateway printf "a downlink was already scheduled, overwritting it...".gateway  useed  the software simulated spi with mode 0,would u pls tell me how to solve this problem ,thinks very much!!!