Lora-net/lora_gateway #162: Upgrade to kernel 5.4.51-v7+ breaks Raspberry Pi Hats that use SPI and expect GPIO 7 to be free

ludionisio, 2020-07-24:
A recent update of the kernel of Raspberry Pis changed the Device Tree Overlay setting for the SPI and enabled by default the Chip Select Pins 7 and 8.

The shown error is:
`ERROR: Failed to load fw 1`
`ERROR: Version of calibration firmware not expected, actual:0 expected:2`
`ERROR: [main] failed to start the concentrator`

In order to continue using these Raspberry Pi Hats that follow the reference design, which use GPIO 7 for resetting the concentrator it is now necessary to add the following line to the `/boot/config.txt`:

`dtoverlay=spi0-cs,cs1_pin=25`

This directive reroutes Chip Select 1 to GPIO 25, you may also use another free GPIO if your setup requires so.

closed 2020-07-24 by ludionisio

sp193, 2020-12-01:
@ludionisio's post helped me to compile libloragw and run the UDP Packet Forwarder with a modern Raspbian image. Thanks.
The complete discussion was here at raspberrypi/linux#3749.
I would like to add on, with some of my own observations.

The new kernel will claim GPIO7 (SPI0 CS1) if SPI is enabled, but IMO this only really blocks the reset procedure. Communication with the board may fail, due to it not getting reset. If ``./reset_lgw.sh start`` is executed:
```
pi@raspberrypi:~/lora_gateway $ ./reset_lgw.sh start
Accessing concentrator reset pin through GPIO7...
./reset_lgw.sh: 27: echo: echo: I/O error
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 34: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 37: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
```

The "I/O Error" is actually EBUSY (Device or resource busy), due to the kernel already reserving GPIO7 and making it impossible to export GPIO7 in ``/sys/class/gpio``. A possible solution would be to remap at least CS**1** of SPI0, by adding an overlay to ``/boot/config.txt``.

I also found that it is important to keep CS0 as GPIO8, as the reference board (and compatibles) uses this. libloragw accesses ``/dev/spidev0.0`` by default, which refers to CS0 of SPI0.
If the board is not enabled, reads over SPI may result in 0s getting read.

The Semtech starter kit also uses some GPIO pins for the LCD panel. From ``/opt/lora-net/misc/lcd.py``:
```
# Raspberry Pi pin configuration:
lcd_rs        = 25  # original 27  # Note this might need to be changed to 21 flor older revision Pi's.
lcd_en        = 24  # original 22
lcd_d4        = 23  # original 25
lcd_d5        = 17  # original 24
lcd_d6        = 27  # original 23
lcd_d7        = 22  # original 18
lcd_backlight = 4
```
I believe it is better to use GPIO5 instead of 25.

rbaldwin13, 2021-06-25:
In an effort to improve our customer support experience and in recognition that our support backlog on GitHub has historically exceeded the capacity of our engineering team, we have taken the difficult decision to focus on the most contemporary issues reported and to close all others without confirmation of resolution.

Our belief is that issues which have remained unresolved and unaltered for extended periods of time are less likely to continue to pose a significant problem to the user than when they were originally filed. More contemporary issues however may still be relevant and hence are more appropriate to prioritize.

For those users who remain interested in resolution of a reported issue that was closed, we are encouraging usage of our developer portal forums [https://forum.lora-developers.semtech.com/] and commercial support portal [https://semtech.force.com/ldp/ldp_support?__hstc=212684107.579a13689e43099691e328c9248e6ecc.1623103335314.1624470656297.1624635514661.6&__hssc=212684107.6.1624635514661&__hsfp=4176385453] as the preferred avenues to receive support. We will continue to monitor the GitHub issue trackers as well, but want to encourage all users to take advantage of the increased community presence on the developer portal. For commercial customers, we highly recommend using the commercial support portal which is uniquely tailored to service such support requests.

gatismu, 2024-03-10:
> @ludionisio's post helped me to compile libloragw and run the UDP Packet Forwarder with a modern Raspbian image. Thanks. The complete discussion was here at [...]

Hello!
if I may ask.. is there fix for that? 
I also updaer rpi4, in with theory that external SSD constantly drops out.. but now my LoRa gateway is " broken" 

`$ ./reset_lgw.sh start
./reset_lgw.sh: 31: echo: echo: I/O error
./reset_lgw.sh: 32: echo: echo: I/O error
./reset_lgw.sh: 33: echo: echo: I/O error
./reset_lgw.sh: 34: echo: echo: I/O error
./reset_lgw.sh: 37: cannot create /sys/class/gpio/gpio17/direction: Directory nonexistent
./reset_lgw.sh: 38: cannot create /sys/class/gpio/gpio5/direction: Directory nonexistent
./reset_lgw.sh: 39: cannot create /sys/class/gpio/gpio18/direction: Directory nonexistent
./reset_lgw.sh: 40: cannot create /sys/class/gpio/gpio13/direction: Directory nonexistent
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
./reset_lgw.sh: 50: cannot create /sys/class/gpio/gpio18/value: Directory nonexistent
./reset_lgw.sh: 52: cannot create /sys/class/gpio/gpio17/value: Directory nonexistent
./reset_lgw.sh: 53: cannot create /sys/class/gpio/gpio17/value: Directory nonexistent
./reset_lgw.sh: 55: cannot create /sys/class/gpio/gpio5/value: Directory nonexistent
./reset_lgw.sh: 56: cannot create /sys/class/gpio/gpio5/value: Directory nonexistent
./reset_lgw.sh: 58: cannot create /sys/class/gpio/gpio13/value: Directory nonexistent
./reset_lgw.sh: 59: cannot create /sys/class/gpio/gpio13/value: Directory nonexistent
`
it there easy fix fort this?

`Linux m1box 6.6.21-v8+ #1741 SMP PREEMPT Fri Mar  8 13:40:00 GMT 2024 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Mar 10 12:58:32 2024 from 192.168.1.2
`
Thanks!