Lora-net/lora_gateway #184: Documentation about HAL library

Rynxie, 2024-07-12:
Hi, I am researching LoRa gateways and I want to try building one. But Semtech's SX13xx chips doesn't have a documentation of registers or HAL library itself. How can I adapt this HAL library to MCU. Can you provide more detailed documentation about this? Or is there a different chip that have documented registers for gateways?

mcoracin, 2024-07-12:
Hello,

First of all, if you build a gateway nowadays, I guess that you will rather use sx1302/1303 chip, not sx1301, right ?
So you need to use the sx1302_hal repository as reference.

Then, there is no documents more than the sx130x datasheet that you can find on Semtech's website. But for sure it will not allow you to build a gateway without reusing the HAL we provide.
It is a complex chip to configure, so we provide reference code instead of documentation, to be used "as is".

Of course you can port it on any MCU, by adapting the SPI, GPIO, I2C handling code.

Best regards,
Michael

embed-engineer, 2024-07-12:
Hi Tunahan,

Trying to implement a lora gateway with current resources on a MCU is very hard work. You need to research on highend processors which support openwrt. shortly, research key words that you need are openwrt, chirpstack, sx1302 hal library... 

Ahmet

mcoracin, 2024-07-12:
Hi @embed-engineer , I would say that it depends on the targeted MCU. For example implementing it on a modern ESP32 (like the S3) is quite straight forward. Also, of course, any linux capable platform is even more simple as the sx1302_hal project can almost be used "as is".

Rynxie, 2024-07-12:
Thank you guys for answers. The little thing I don't understand is how to use this code. I mean it is literally print out the data to standard I/O. Where is the SPI frame sending part etc. I think I missed the SPI frame sending so I will appricated if you guys point it to me. I will look into the keywords that Ahmet told me and will look into the SX1302 HAL library.

mcoracin, 2024-07-12:
@Rynxie you should start by reading the readme.md file here:
https://github.com/Lora-net/sx1302_hal/tree/master/libloragw

And maybe run the whole project on a raspberry pi host, with a sx1302 hat (seeed, waveshare, ...).
This will help you understand how things work.

Please raise questions on the sx1302_hal project if needed.

Rynxie, 2024-07-12:
@mcoracin  Thanks a lot. I will ask some questions probably :smile:

closed 2024-07-12 by Rynxie