Lora-net/lora_gateway #185: Difference between util_spectral_scan and the legacy util_band_survey

sp193, 2024-08-04:
Hi.

What's the difference in between util_spectral_scan and [util_band_survey](https://github.com/Lora-net/lora_gateway/tree/45be5ff473ad1f39829bfcebc641201ae46c1c80/util_band_survey)? Sometime around 3.0.0, util_band_survey was removed without any explanation. From the documentation available, it doesn't seem to require a FPGA, so it seems to be usable for the older/basic SX1301 gateways.

Does this (older) design have any notable limitations, compared to the new util_spectral_scan tool that involves the FPGA?

I suppose I ask out of curiosity, since there's no documentation on the inner workings of the SX1301. Why/why not some things are done.

Thanks in advance.

mcoracin, 2024-08-06:
Hi,
To be honest we have no history about this util_band_survey utility, so we won't be able to support you.
Still, you can try it and see it it does what you need.

The util_spectral_scan utility provides the ability to perform the spectral scan in background, while the sx1301 operates as a LoRa Gateway. That's why it requires the FPGA and an extra radio.

Unfortunately, I cannot give you more details about possible limitations with util_band_survey due to lack of history.

Best regards,
Michael

sp193, 2024-08-07:
Hi, hello.

I understand. Thank you for taking the time to explain.
If I somehow find this tool very useful, I will mention what I find.